NextThingCo/CHIP-linux #7: Midi support is missing in latest production images

DaveCasey, 2016-03-10:
I'm not sure if this is the correct repository for reporting this, but it looks like midi support stopped working in the production images somewhere between [the hackster.io midi article](https://www.hackster.io/11802/c-h-i-p-midi-arpeggiating-synth-e311ab) and when the production boards shipped.
```
chip@chip:~$ amidi -l
Dir Device    Name
IO  hw:1,0,0  USB O2 MIDI 1
chip@chip:~$ aplaymidi -l
ALSA lib seq_hw.c:457:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
Cannot open sequencer - No such file or directory
```

After digging through the bbs (You folks have created a really great and supportive community, btw.) I tried grabbing one of Renzo's kernel images, and it seems to resolve the issue. This [thread](https://bbs.nextthing.co/t/how-do-i-configure-bristol-soft-synth-on-c-h-i-p/2594/4) suggests that the solution involves:

`make prepare scripts
make menuconfig #Navigate the menu to enable snd_seq and snd_seq_midi as modules
make modules SUBDIRS=sound`

-but rebuilding kernels and/or parts of them is a bit out of my league for now.

Thanks,
-Dave

jasonabele, 2016-03-11:
Dave, this is on our radar for an upcoming release and I have a proposed branch with the kernel config changes available here: https://github.com/NextThingCo/CHIP-linux/tree/ja/4.4/add_midi

Once someone with a little more midi experience tells me that those are sufficient changes, we'll merge it and eventually there will be new kernel deb's with midi support to apt-get.

If you are using buildroot, you could just change your config to use the ja/4.4/add_midi branch and let me know if it works for you.

seclorum, 2016-05-11:
I'm here for all MIDI testing, although I don't have a CHIP, I do have near-infinite MIDI.  If I order a CHIP today, will it get here soon, or should I rather wait for my PocketCHIP to arrive?