NextThingCo/CHIP-linux #10: R8 GPIO Edge Detection

xtacocorex, 2016-04-02:
When exporting GPIO that are hooked directly to the R8, there is no way to set edge detection via the sysfs interface.

Edge detection is supported via the XIO GPIO on the CHIP.

If I missed a commit that enables edge detection for R8 based GPIO, please close this issue.

mripard, 2016-04-08:
Hi,

Which pin are you using? Only a limited number of GPIOs can be used as an interrupt source on the R8, so it might be possible that you were using one that can't?

xtacocorex, 2016-04-08:
I was trying to use CSID0.  Didn't think about that, are the 2 that can be used AP-EINT3 and AP-EINT1?

xtacocorex, 2016-04-08:
Looks like AP-EINT1 and AP-EINT3 offer edge detection.  Will close this issue as it's now OBE.

closed 2016-04-08 by xtacocorex