NextThingCo/CHIP-linux #14: Add ENC28J60 support for HUMMUS DIP

zerotri, 2016-06-20:
As per this thread: https://bbs.nextthing.co/t/ethernet-shield-hummus-dip/3516
CHIP community member JKW has created a DIP to add ethernet over SPI. CONFIG_ENC28J60 should be enabled as a module.

zerotri, 2016-06-20:
This is to be left open until I can get in contact with community member JKW concerning what is still to be done to support the HUMMUS DIP

KoljaWindeler, 2016-07-13:
Hi, besides adding the enc28j60 module you also have to apply renzos patch  http://www.raspibo.org/renzo/spi-sun4i.patch

That fixes two issues:
1. Spi Speed 
https://bbs.nextthing.co/t/flashing-atmega-chips-over-xio-gpio-spi-with-avrdude/2955/8?u=jkw
2. Spi FIFO size limitiation
http://lists.infradead.org/pipermail/linux-arm-kernel/2014-April/244305.html

Thanks JKW