NextThingCo/CHIP-linux #20: more filesystem modules

docjkl, 2016-06-29:
Hi, 

CHIP's kernel currently lacks exfat and hfs+ support, so flash drives formatted with either Windows or OS X, respectively, will not work.   I believe that they should be included as part of the standard kernel build for maximum Interoperability.

insanemal, 2016-06-30:
I'll actually add it here as this is probably the right place,

Can we also get XFS and CephFS.

I use XFS on many things so having them just work when plugged in via USB would be great. 

I also have a Ceph cluster that I use in place of a NAS, so getting access to my CephFS would be a huge plus.

tchabaud, 2016-07-04:
Hi,
CIFS support would also be great. I want my Chip to be able to read content from my NAS.

insanemal, 2016-07-04:
Yeah CIFS would also be good... 

Really, most modules should be enabled in M form... there is no good reason (including space) to not do this.

dos1, 2016-10-08:
How is it that so simple issue is ignored for such a long time?
Adding modules, especially the most requested ones, is a no-brainer and doesn't cost anything for anyone not using them. You can even leave them out of default image, just provide matching modules to the prebuilt kernel to download.

It's ridiculous to have to compile kernel on my own if I want to use btrfs or docker...