NextThingCo/CHIP-linux #27: CP2102 USB-to-Serial Driver should be added to version 4.4

Flexshot, 2016-07-06:
```
on my Pi, when I plug in the data logger, I see this:

Jun 30 22:59:28 rpi kernel: [462649.390492] usb 1-1.4: new full-speed USB device number 5 using dwc_otg
Jun 30 22:59:28 rpi kernel: [462649.498217] usb 1-1.4: New USB device found, idVendor=10c4, idProduct=ea60
Jun 30 22:59:28 rpi kernel: [462649.498239] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 30 22:59:28 rpi kernel: [462649.498252] usb 1-1.4: Product: CP2102 USB to UART Bridge Controller
Jun 30 22:59:28 rpi kernel: [462649.498265] usb 1-1.4: Manufacturer: Silicon Labs
Jun 30 22:59:28 rpi kernel: [462649.498277] usb 1-1.4: SerialNumber: 0001
Jun 30 22:59:28 rpi kernel: [462649.504888] cp210x 1-1.4:1.0: cp210x converter detected
Jun 30 22:59:28 rpi kernel: [462649.505552] usb 1-1.4: cp210x converter now attached to ttyUSB0
Jun 30 22:59:28 rpi mtp-probe: checking bus 1, device 5: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4"
Jun 30 22:59:28 rpi mtp-probe: bus: 1, device: 5 was not an MTP device

When I plug it into the CHIP (either directly or via a powered USB hub), I only get this:

Jun 30 23:26:50 dcrweather kernel: [ 1138.330000] usb 1-1.3: new full-speed USB device number 9 using ehci-platform
Jun 30 23:26:50 dcrweather mtp-probe: checking bus 1, device 9: "/sys/devices/platform/soc@01c00000/1c14000.usb/usb1/1-1/1-1.3"
Jun 30 23:26:50 dcrweather mtp-probe: bus: 1, device: 9 was not an MTP device
```
-- https://bbs.nextthing.co/t/cant-get-a-dev-tty-for-a-vantage-vue-usb-data-logger/5844

ldfm, 2016-11-19:
Is there a chance this will make the next update? It's the last thing keeping me from using the CHIP as a weewx weather station...

ldfm, 2016-12-22:
This appears addressed in the last 4.4 update...