NextThingCo/CHIP-linux #30: gpio group with /sys/class/gpio access (like i2c group)

ugate, 2016-07-21:
Grant the default chip user with access to `/sys/class/gpio` with a new `gpio` group like `i2c` group that chip is added to prevent elevated privilege access using `sudo`.

mripard, 2016-07-26:
Opening up /dev/mem access to all users is an enormous security hole.

And I'm guessing that if you need that and mention GPIO access, you're doing something that's not very right to me.

ugate, 2016-07-26:
@mripard That's why `dev/gpiomem` was mentioned. You don't have to open up everything in `/dev/mem`, but rather just the stuff relevant to GPIO. Look at how Raspberry Pi has already done it: raspberrypi/linux#1112

mripard, 2016-09-15:
What is the problem exactly? We're discussing solutions to an unstated problem, which doesn't really help.

/dev/gpiomem is not the solution anyway.
It brings a whole new set of problems and hazards ranging from frying your board entirely to various driver crashes.

You would also reimplement a driver that has been widely tested already, and for no particular benefits, since you can't leverage the raspberry pi code anyway.

Flexshot, 2016-09-16:
Hey @mripard, 
What solution do you recommend to solve this issue for our users?

mripard, 2016-09-16:
@Flexshot You still haven't said what the issue is

Flexshot, 2016-09-20:
requiring a user to have root privileges to access GPIO

mripard, 2016-09-27:
Have you looked at /sys/class/gpio?