NextThingCo/CHIP-linux #32: More modules

TobbeJ, 2016-07-27:
Please consider enabling more drivers as modules instead of leaving them out completely.
Things on my wish list is usb drivers for dvb cards and usb based web cams.

i compiled my own kernel with dvb drivers and got it to work, same with webcam drives.
i would assume others are interested in webcams too, for example for robotics projects or similar with webcams involved.

smiller171, 2016-08-20:
:+1: This isn't an embedded system. You should compile with as many modules as you can and let the user enable them at will, or at least provide a separate image with a more complete kernel. I am having to jump through hoops to make a CHIP Docker Swarm.

joostliketoast, 2016-09-02:
@smiller171 Did you manage to get docker running on CHIP? if so, care to share how you managed?

smiller171, 2016-09-02:
@joostliketoast I actually did. I intend on doing a writeup soon, but shoot me an email or find me on Gitter and I'll give you the basics. My email is available in my profile.

combs, 2016-11-26:
@mripard any chance of this making the next release? 🎱  😄

akatrevorjay, 2017-01-02:
Now I have to cross compile a kernel just to get usbip to work.
Please follow the standard ubuntu/debian kernel guide.
It will literally walk you through the process, and provide you with a decent starting config to boot.

akatrevorjay, 2017-01-02:
The more I think about it, the more I wonder why NTC has to distribute their own kernel at all? Are platform related patched required? If so, are they submitted upstream?