NextThingCo/CHIP-linux #33: Add CONFIG_WIRELESS_EXT to the kernel

hadess, 2016-07-28:
So people can use the "upstream-ish" version of the driver, or at least test it.

It currently fails to build with:
```
/home/chip/rtl8723bs/os_dep/os_intfs.c: In function ‘rtw_init_netdev’:
/home/chip/rtl8723bs/os_dep/os_intfs.c:567:9: error: ‘struct net_device’ has no member named ‘wireless_handlers’
  pnetdev->wireless_handlers = (struct iw_handler_def *)&rtw_handlers_def;
         ^
```

jdub, 2016-11-12:
Nice to see you here! Concur, it's a bummer that a full build is required to enjoy the upstream-ish driver. I'm testing a build now to see how it goes.

jasonabele, 2016-11-18:
I am missing the details here, but I believe the release kernel builds from NTC all have CONFIG_WIRELESS_EXT, can you give me more context about which kernel you are using?

jdub, 2016-11-18:
4.4.13-ntc-mlc from the most recent images. Definitely no CONFIG_WIRELESS_EXT.