NextThingCo/CHIP-linux #40: Enable TUN/TAP device driver (CONFIG_TUN)

HeatfanJohn, 2016-08-05:
Enabling TUN/TAP device driver (CONFIG_TUN) is required to use OpenVPN.

> 
> TUN/TAP provides packet reception and transmission for user space
> programs. It can be viewed as a simple Point-to-Point or Ethernet
> device, which instead of receiving packets from a physical media,
> receives them from user space program and instead of sending packets
> via physical media writes them to the user space program.
>   
> When a program opens /dev/net/tun, driver creates and registers
> corresponding net device tunX or tapX. After a program closed above
> devices, driver will automatically delete tunXX or tapXX device and
> all routes corresponding to it.

Kernel config needs:

    CONFIG_TUN=m

computermouth, 2016-11-29:
Done!

closed 2016-11-29 by computermouth

kragil, 2017-08-14:
Which kernel do I need to install to get this?