NextThingCo/CHIP-linux #45: Early device tree overlay limitation

KoljaWindeler, 2016-08-16:
Set device tree overlay during u-Boot only for ntc dips. Otherwise only the last overlay will be loaded. Stacking two boards will load the overlay for the board with higher eeprom Id which might be wrong.

Please have a look right here:
https://bbs.nextthing.co/t/dip-device-autoload-on-boot-no-working/7671/18?u=jkw

mripard, 2016-08-25:
No, this is by design.

We want to allow other people if relevant to be able to put their own DT here and have U-Boot apply it.

In your case, your issue is that you placed that overlay in early, while you really shouldn't.

KoljaWindeler, 2016-08-25:
The current uBoot design is very poor, please have a look at 
https://github.com/NextThingCo/CHIP-u-boot/blob/nextthing/2016.01/next/board/sunxi/dip.c#L122

There is a look over all devices on the bus, 

for (device_find_first_child(bus, &dev); dev;
	     device_find_next_child(&dev)) {

but only one string to save the target dio overlay that will be passed on to the kernel
snprintf(dip_name, 64, "dip-%x-%x.dtbo", vid, pid);

therefore only the last DIP in that loop will win and its overlay will be loaded .. 
or in reality: Stack the humus and the vga on 100 chips and you'll end up with 50 video and 50 ethernet enabled chips :) 

I guess the best way would be to fill an array for each file found during the for loop and pass all overlays to the kernel, would you agree? 

Kolja

mripard, 2016-08-26:
You get it wrong. Linux never sees those overlays. They're applied by U-Boot before Linux starts. What Linux sees is just an usual device tree. And really, you shouldn't be using that mechanism for the Hummus.

Back to the main issue. You're the first one ever to test multiple DIPs, so you just encountered a bug, congrats! :)

I just pushed something to our U-Boot that should fix it. Let me know how it goes.

closed 2016-08-26 by mripard

KoljaWindeler, 2016-08-26:
I see, all overlays are applied and the resulting device tree is what the kernel sees.
Thanks for the fix, I'd guess that this fixes the issue.

mripard, 2016-08-26:
Please test, and feel free to reopen the issue if it doesn't.

You're basically the only one that can test right now :)

KoljaWindeler, 2016-08-26:
Honestly, i have no idea how I would update my uboot partition, can you link a guide? 
I'm willing to test

mripard, 2016-08-26:
You don't have to reflash it, you can use FEL!

Start by getting the sunxi-tools https://linux-sunxi.org/Sunxi-tools
Then, put the FEL pin to the ground, download the chip-u-boot repo,

and do https://linux-sunxi.org/FEL/USBBoot#Booting_U-Boot_over_USB, with CHIP_defconfig instead.

You should have your new u-boot starting.

If it works, we'll see how to flash it

KoljaWindeler, 2016-08-26:
Works just fine. This is what I did to compile it http://pastebin.com/mK86EKr0
and this is the boot output http://pastebin.com/8seeJKtz

I've removed my dtbo from the early directory and still have VGA, woop woop.

The only bummer is that I still have to load my overlay on boot with this script, https://github.com/KoljaWindeler/CHIP_HUMUS/blob/master/overlay/load.sh

One version (pChip 4.3? official 4.4?) autoloaded all connected dip overlays whenever I've loaded w1_ds2431 but I can figure out what version that was. My self compiled 4.4.11 doesn't do that, do you know which verseion that was, and will that be a default feature in the next release? 
Thanks

mripard, 2016-08-31:
It's probably in 4.4, and it should be default already.