NextThingCo/CHIP-linux #53: remove 32khz interrupt debounce

tve, 2016-09-03:
Apparently gpio interrupts have a 32khz debounce, which may be useful for certain uses but shouldn't be there for everything. Please remove the debounce or provide a simple way to remove it. See https://bbs.nextthing.co/t/reading-dht11-dht22-am2302-sensors/2383/43 for a patch.

DareDevil73, 2016-12-15:
Has this patch been added to the latest kernel?

foxharp, 2017-01-23:
no, it has not been.  and it's quite an important patch.

i'm having trouble imagining how that debounce clock is supposed to work without a proper description in the user manual. (if anyone can point to one, i'd appreciate it.)

but in any case, applying this patch eliminates the lost interrupt problem i've been pursuing in this forum thread:  https://bbs.nextthing.co/t/neither-poll-nor-epoll-nor-inotify-works-reliably-for-gpio-input/13960/22?u=pgf

interrupts come in from the XIO expander at the same rate as a bouncing switch or button input to the expander.  that can be very fast:  in the low tens of microseconds.  switch closures are often lost completely with the clock at the 32Khz rate, but i haven't observed one lost yet with the 24Mhz patch applied.

attached is a git-format-patch version of the patch, with my sign-off, and author credit to mripard.

paul

[0001-change-the-PIO-interrupt-debounce-clock-from-32KHz-t.patch.txt](https://github.com/NextThingCo/CHIP-linux/files/724412/0001-change-the-PIO-interrupt-debounce-clock-from-32KHz-t.patch.txt)