NextThingCo/CHIP-linux #56: interrupts on XIO pins incorrectly trigger on both edges

tve, 2016-09-11:
I have a device that generates a positive interrupt pulse and I connected that to both XIO-D7 and to AP-EINT3. I'm writing "rising" to the gpio pin's "edge" file in /sys/class/gpio/... I then use epoll to receive the interrupts.

What I observe is that if I use XIO-D7 to generate the interrupt I incorrectly get an interrupt on *both* the rising and falling edge of the signal. If I then change the code to use AP-EINT3 leaving everything else the same I correctly get an interrupt on only the rising edge.

tve, 2016-09-11:
The code that needs fixing to take rising vs falling into account is https://github.com/NextThingCo/CHIP-linux/blob/chip/stable/drivers/gpio/gpio-pcf857x.c#L199-L202

mripard, 2016-09-15:
Unfortunately, I don't have a device to test this on. Do you have the patch you're mentionning somewhere?

tve, 2016-09-15:
I don't have a patch, I was only linking to the code that reads the current pin state and compares to the prior state but fails to take the direction of change into account.

nyboer, 2016-09-16:
@mripard - what do you need to test this? We encountered this a while ago in python, but we thought it was in the python module, not the driver and had to move on.

mripard, 2016-09-16:
@tve Looking more closely at it, it seems like the PCF can only do both rising and falling triggering, but not just a single one. We should be rejecting invalid trigger type though. I'll cook something to do that next week.

tve, 2016-09-16:
Yes, that is correct, you can't tell the PCF which edge you want. It doesn't even tell you which of the 8 pins transitioned. But the driver already checks whether a pin has changed to determine which pin(s) should receive the notification and thus it should be easy to enhance this check to also check the direction of change. I have not had time, the main hurdle for me is figuring out how to communicate to the driver which edge was requested by the ioctl.

mripard, 2016-09-26:
The kernel tries to remain as accurate as possible: if some hardware doesn't implement one feature, it usually doesn't try to emulate it.

In that particular case, building some kind of state machine in the driver would be pretty fragile. If for some reason, one interrupt is missed (spinlock taken, already handling another interrupt), your state machine is entirely off and cannot recover.

The check that is already in place is much more robust in that regard.

tve, 2016-09-29:
> The kernel tries to remain as accurate as possible: if some hardware doesn't implement one feature, it usually doesn't try to emulate it.
> In that particular case, building some kind of state machine in the driver would be pretty fragile

I have to admit that I don't understand your comment. The kernel *already* implements a feature that the hardware doesn't: per-pin interrupts. The hardware provides no way to disambiguate which pin caused the interrupt. The kernel does this by comparing the previously read state with the current state.
I also don't see the need for any state that is additional to the current state. All the kernel needs is to access the edge that was requested and compare it with the current state of the pins that it reads. It's basically just an additional conditional.

mripard, 2016-10-17:
Yes, irq_set_type is the right way to change that.

My point is that the code that you have shown also needs to deal with the fact that you might lose an interrupt (because they were masked for example). In that case, you also need to check that against the value of that pin, for example to avoid notifying a rising front on a falling one, just because you missed the previous one.

I'll try to give it a shot and see how robust it is.

tve, 2016-10-17:
Got it.

I was looking at the code yesterday because I ran into another issue. I wrote a (user-space) test where I connected two of the I/O pins of the chip, set one as input with edge triggered interrupt and the other as output. When I toggle the output I don't get interrupts (epoll). If I connect an output on a completely different port (this was using CHIP) to XIO-D7 and run the same test unchanged the edge interrupts come in just fine. I.e., connecting XIO-D6 to XIO-D7 doesn't work, same user space code connecting PWM1 (gpio34) to XIO-D7 works.

I need to troubleshoot further to make sure it's really a driver problem as opposed to some user-space problem but looking through the driver code I noticed that gpio->status does not get initialized when edge detection / interrupt is turned on on a specific pin. Unless I'm missing something, it only gets initialized when the driver is initially loaded and the device is detected. This means that whether the first edge is detected by the `gpio->status ^ status` expression is somewhat random.

mripard, 2016-11-07:
Sorry for getting back to you on this so late.

Did you make any progress on this? I'm guessing having the status initialised properly would be the very first step into building that robust state machine.

tve, 2016-11-07:
No progress on my end. I'm working around at user-level. Would just be nice to fix this eventually given that it's in every CHIP...

xtacocorex, 2016-12-31:
This explains my hack in CHIP_IO to combat this issue.  Totally didn't realize it was something that could be fixed in the kernel.

foxharp, 2017-01-14:
I'm happy to work on this if need be -- looks like most of the heavy listing has been done, and it just needs to be all pulled together, and tested.  If someone's already on it, please let me know, and I'll leave it be.

foxharp, 2017-01-23:
After playing with an implementation of the fixes discussed in this thread, I have to say that I don't really recommend implementing the rising/falling change.  Bouncing switches can cause rising or falling interrupts when both opening and closing, which means that the true state of the input pin will need to be checked at user level in any case.  It's probably better that the user quickly discover that they're getting notification on both edges, and let them deal with it, than to try and do the filtering and get it wrong some small percent of the time.

I do have a clean patch which implements the change, though, if anyone wants it.

I also have a separate patch which adds the reinitialization of the input port status when interrupts are enabled, to avoid false initial conditions.

foxharp, 2017-01-23:
here are the two patches:
[0001-initialize-current-port-status-when-enabling-interru.patch.txt](https://github.com/NextThingCo/CHIP-linux/files/725001/0001-initialize-current-port-status-when-enabling-interru.patch.txt)
[0002-let-user-request-just-rising-or-falling-edge-interru.patch.txt](https://github.com/NextThingCo/CHIP-linux/files/725002/0002-let-user-request-just-rising-or-falling-edge-interru.patch.txt)