NextThingCo/CHIP-linux #64: Built-in support for NAT / IP tables to make CHIP as wifi repeater

vikrant8051, 2016-11-16:
There are two Wi-Fi interfaces but they are useless without built in support for NAT/IP table. 
Please add support for it in kernel. It will help to develop CHIP as Wi-Fi repeater.

It is basic thing. All embedded  board provides built in support for NAT/Ip table for eg, Raspberry Pi.

I've already enable wi-fi Access point on CHIP using hostapd but can't execute 

iptables -t nat -A POSTROUTING -o wlan0 -j MASQUERADE
iptables -A FORWARD -i wlan0 -o wlan1 -m state --state RELATED,ESTABLISHED -j ACCEPT
iptables -A FORWARD -i wlan1 -o wlan0 -j ACCEPT

these commands to link wlan0 & wlan1.

Building kernel is very tedious process & I don't know which parameter in kernel .config file should enable to execute above commands. Please !! 

Thanks in advanced !!

ldfm, 2016-11-19:
It also appears iptables needs to be build in in order to use fail2ban:

2016-11-18 16:10:07,257 fail2ban.actions.action[18482]: ERROR   iptables -N fail2ban-ssh
iptables -A fail2ban-ssh -j RETURN
iptables -I INPUT -p tcp -m multiport --dports ssh -j fail2ban-ssh returned 300

~ $ iptables -L
modprobe: ERROR: ../libkmod/libkmod.c:557 kmod_search_moddep() could not open moddep file '/lib/modules/4.1.19-v7+/modules.dep.bin'
iptables v1.4.21: can't initialize iptables table `filter': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.

danrl, 2016-12-08:
This is so basic that I am stunned it isn't part of the default image.

mikecentiq, 2016-12-13:
Does the new kernel they've just released include this?