NextThingCo/CHIP-linux #73: ALSA Causing Stutter/Hanging in Pico-8, Mednafen, Sunvox, LGPT, PocketC.H.I.P.

mackemint, 2017-01-04:
There are frequent audio dropouts that all seem to relate to buffer underrun in Alsa. Using an external USB sound card with settings in .asoundrc  buffer_size  to 8096 and period size to 221 seems to dramatically reduce audio dropouts in Sunvox according to my own research, but it still outputs these error messages:

ALSA snd_pcm_writei error: Broken pipe
ALSA EPIPE (underrun)

mackemint, 2017-01-11:
More information and research here:
https://bbs.nextthing.co/t/alsa-causing-stutter-hanging-in-pico-8-mednafen-retroarch-arm-solved-maybe-please-help-us-test-verify/13765

mackemint, 2017-02-01:
Some improvement can be achieved  by disabling several services that are running as CGROUPS at the kernel level to do things like determine battery charge etc. 

Disabling These services makes some improvement :
https://github.com/NextThingCo/pocketchip-batt/tree/master/pocketchip-batt/system

pocketlight, 2017-03-25:
This causes stutter/hangs in "all" (at least the ones I've tested) apps that uses audio. Most noticeable in openArena, openTTD.