NextThingCo/CHIP-linux #78: nextthing/4.9/chip clone isn't successful

gitkobold, 2017-03-24:
git clone https://github.com/NextThingCo/CHIP-linux.git --single-branch --branch nextthing/4.9/chip --depth 1
Cloning into 'CHIP-linux'...
remote: Counting objects: 59706, done.
remote: Compressing objects: 100% (56302/56302), done.
Receiving objects: 100% (59706/59706), 156.79 MiB | 2.05 MiB/s, done.used 0eceiving objects: 100% (59706/59706), 156.70 MiB | 1.98 MiB/s

Resolving deltas: 100% (5089/5089), done.
Checking connectivity... done.
**error: unable to create file drivers/gpu/drm/nouveau/nvkm/subdev/i2c/aux.c (No such file or directory)
error: unable to create file drivers/gpu/drm/nouveau/nvkm/subdev/i2c/aux.h (No such file or directory)**
Checking out files: 100% (56235/56235), done.
fatal: unable to checkout working tree
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry the checkout with 'git checkout -f HEAD'

combs, 2017-03-24:
hmm, I don't know if this is helpful or not but I was able to clone it just now on my machine:

```
Darwin hostname.domain.com 14.5.0 Darwin Kernel Version 14.5.0: Sun Sep 25 22:07:15 PDT 2016; root:xnu-2782.50.9~1/RELEASE_X86_64 x86_64
```

```
hostname:~ me$ cd /tmp
hostname:tmp me$ git clone https://github.com/NextThingCo/CHIP-linux.git --single-branch --branch nextthing/4.9/chip --depth 1
Cloning into 'CHIP-linux'...
remote: Counting objects: 59706, done.
remote: Compressing objects: 100% (56302/56302), done.
remote: Total 59706 (delta 5089), reused 19128 (delta 2514), pack-reused 0
Receiving objects: 100% (59706/59706), 156.79 MiB | 7.85 MiB/s, done.
Resolving deltas: 100% (5089/5089), done.
Checking connectivity... done.
Checking out files: 100% (56235/56235), done.
```

gitkobold, 2017-03-24:
Interesting. I've tried it more times but always get this error :-o
Any idea?
Btw, do U know this 4.9 is working?

tgies, 2017-08-04:
Yes, [Windows doesn't want you to create a file called 'aux'](https://superuser.com/questions/613313/why-cant-we-make-con-prn-null-folder-in-windows) (or, for that matter, aux.*)